openthread/openthread #10938: The otRadioFrame->mInfo.mTxInfo.mCslPresent field is no longer updated correctly

marius-preda, 2024-11-18:
After #10692, the field otRadioFrame->mInfo.mTxInfo.mCslPresent is no longer updated. 
NXP platforms were using that flag to know when to updated the CSL IE.  One solution is to use the value of the csl period( >0 ) to know when to update the CSL IE but I consider this solution not perfect. We need to clarify what is the intent, if this fileld is no longer supported then it should be removed from the openthread interface. If the field is sitll relevant, as I consider it, then a fix should be made to update it again.

abtink, 2024-11-18:
Thanks @marius-preda for resporting this.

I think we should fix this in the code so `mCslPresent ` is set when CSL IE is present. I can submit a quick fix PR for this.

abtink, 2024-11-18:
Submitted #10939

jwhui, 2024-11-19:
@marius-preda , thanks for raising this issue!

PR #10939 (2024-11-19): [mac-frame] set `mCslPresent` on `TxFrame` when CSL IE is added
This commit ensures that the `mCslPresent` flag is set on `TxFrame` when the CSL IE header is appended to the frame. This addresses an issue introduced in #10692 where this flag can remain unset.

---

Resolves #10938