openthread/openthread #10971: When child on sleepy mode changes to non-sleepy, the messages in the send queue destined to the child are discarded by parent.

jinpeng1989, 2024-11-26:
When the child device switches from rx-off-when-idle to rx-on-when-idle, the parent discards the messages destined for the child device. Unexpected packet lost always occur in this scenario.

The delay-aware queue management is enabled. A direct message that is not ECN capable will be discarded if it is in the send queue for more than kTimeInQueueMarkEcn (default is 500) milliseconds. When child on sleepy mode changes to non-sleepy, the indirect messages in the send queue will be set to direct messages. These messages may have been in the queue for more than kTimeInQueueMarkEcn milliseconds, so they are discarded immediately. 

https://github.com/openthread/openthread/blob/main/src/core/thread/indirect_sender.cpp#L200

https://github.com/openthread/openthread/blob/main/src/core/thread/mesh_forwarder.cpp#L238-L241

jinpeng1989, 2024-11-26:
One possible fix is to set the timestamp of messages to now when it is set to direct messages.

abtink, 2024-11-27:
Submitted https://github.com/openthread/openthread/pull/10979 for this.

jinpeng1989, 2024-11-28:
Thank you very much. We will take this fix and give you feedback after testing.

jinpeng1989, 2024-11-28:
I have one more question. The product that has this issue is a [esp-thread-br](https://github.com/espressif/esp-thread-br), let's call it Product A. Another product (Product B) is an [ot-br-posix](https://github.com/openthread/ot-br-posix), in this scenario, the message is not discarded. After switching to non-sleepy, the message is immediately sent directly to the child. My question is why the two products behave differently.
[image]

jinpeng1989, 2024-11-28:
Product A is based on [ESP-IDF commit dc859c1](https://github.com/espressif/esp-idf/tree/dc859c1e67e0d3b48252a907b89e414cc84fedbb), the corresponding OpenThread commit is [be7d36e ](https://github.com/espressif/openthread/tree/be7d36e4ff9cf7df6dfce54e58a31163c87b93f7).
Product B is based on [Gecko SDK 4.4.3](https://github.com/SiliconLabs/gecko_sdk/releases/tag/v4.4.3), the corresponding OpenThread commit is [7074a43e4 ](https://github.com/openthread/openthread/tree/7074a43e4).

jinpeng1989, 2024-11-28:
Both of these products enable delay-aware queue management. In this scenario, why do they behave differently?

jinpeng1989, 2024-11-28:
> Submitted #10979 for this.

After incorporate this fix on esp-thread-br. The test result is that the packet was not lost after switching to non-sleepy.

closed 2024-12-02 by jwhui