openthread/openthread #10975: Sleepy end device wakes up more often than every poll period

kkasperczyk-no, 2024-11-27:
**Describe the bug** 
The sleepy end device with relatively big poll period (e.g. 600 s) wakes up more often than every assumed 600 s.
The reason is that there are at least two parameters resulting in waking up the device earlier:

* OPENTHREAD_CONFIG_CHILD_SUPERVISION_CHECK_TIMEOUT (by default 190 s)
* OPENTHREAD_CONFIG_MLE_CHILD_TIMEOUT_DEFAULT (by default 240 s)

In such scenario the SED is going to wake up after 190 s due to the child supervision check timeout.

**Expected behavior**

That the SED is going to wake up exactly after the selected poll period value.

**Additional context**
Question: is this behavior expected? Wouldn't it make sense to set the OPENTHREAD_CONFIG_CHILD_SUPERVISION_CHECK_TIMEOUT and OPENTHREAD_CONFIG_MLE_CHILD_TIMEOUT_DEFAULT to the values equal or greater than poll period? As the poll period may change at runtime, would it make sense to automatically update these two parameters if poll period changes?

abtink, 2024-11-27:
> Question: is this behavior expected?

Yes.
I would consider the situation you described a configuration issue.

> Wouldn't it make sense to set the `OPENTHREAD_CONFIG_CHILD_SUPERVISION_CHECK_TIMEOUT` and `OPENTHREAD_CONFIG_MLE_CHILD_TIMEOUT_DEFAULT` to values equal to or greater than the poll period?

Not necessarily. A device should be able to choose any set of parameters it wants. As an example, a device may be using CSL as its indirect transmission mechanism (along with a long data poll transmission period, which would then be longer than the configured supervision timeout).

The OT stack provides APIs to control the "supervision" check timeout and interval at run-time (in addition to the default constants). Note that changing the interval would trigger a "child update" exchange with the parent, as it needs to be synchronized (this is similar to changing the "child timeout").

> As the poll period may change at runtime, would it make sense to automatically update these two parameters if the poll period changes?

This should really be controlled per device (based on its needs and expected traffic). I don't think there is a proper common way to say how the supervision "check timeout" should be mapped to the current "poll period" (or CSL period, etc.).

We have the same situation with "child timeout," which also impacts how often a sleepy end device (SED) may send a data poll. For example, if the "child timeout" is set to 500 seconds, then the device will send polls every 500 seconds (minimum), even if the "poll period" is set to a higher value. Changing the poll period will not update the "child timeout."

Also, note that data poll transmission can be directly controlled by the next layer (using the `otLinkSendDataRequest()` API to send whenever it wants). In this situation, there may be no set "poll period" value using `otLinkSetPollPeriod()`.

kkasperczyk-no, 2024-11-28:
@abtink thanks for all clarifications. It's clear to me now. I'm closing this one, as it seems there is nothing to be fixed beside our local configuration.

closed 2024-11-28 by kkasperczyk-no