openthread/openthread #10983: Drop Udp Messages with Unknown Issues

aaa147147, 2024-12-03:
[image]

aaa147147, 2024-12-03:
[959.log](https://github.com/user-attachments/files/17988525/959.log)
Please find attached our complete test log

aaa147147, 2024-12-03:
We are using the latest version of OTBR

abtink, 2024-12-03:
Messages are dropped if the destination is unknown or unreachable:

1.  The destination IPv6 address is an on-mesh address, but no device on the mesh claims to have this address (there is no response to an "Address Query" for the address).
2.  There is no route toward the intended destination.

The logs provide the answer in your case, as there seems to be no response to the Address Query earlier:

```
> cat ~/Downloads/959.log | grep otbr-agent | grep fd9c:1c81:262e:1:67f6:b2f6:c04b:2a48 -B 2

12-02 06:08:49.819 I/otbr-agent( 1643): 00:01:34.254 [I] AddrResolver--: Timed out waiting for AddressNotify for fd9c:1c81:262e:1:67f6:b2f6:c04b:2a48, retry: 15
12-02 06:08:49.819 I/otbr-agent( 1643): 00:01:34.254 [N] MeshForwarder-: Dropping IPv6 UDP msg, len:158, chksum:0baf, ecn:no, sec:yes, error:AddressQuery, prio:low
12-02 06:08:49.819 I/otbr-agent( 1643): 00:01:34.254 [N] MeshForwarder-:     src:[fd9c:1c81:262e:1:133b:c363:a00d:8bbf]:32992
12-02 06:08:49.819 I/otbr-agent( 1643): 00:01:34.254 [N] MeshForwarder-:     dst:[fd9c:1c81:262e:1:67f6:b2f6:c04b:2a48]:33521
--
...
--
12-02 06:08:59.436 I/otbr-agent( 1643): 00:01:43.872 [N] MeshForwarder-: Dropping IPv6 UDP msg, len:158, chksum:e028, ecn:no, sec:yes, error:Drop, prio:low
12-02 06:08:59.436 I/otbr-agent( 1643): 00:01:43.872 [N] MeshForwarder-:     src:[fd9c:1c81:262e:1:133b:c363:a00d:8bbf]:44134
12-02 06:08:59.436 I/otbr-agent( 1643): 00:01:43.872 [N] MeshForwarder-:     dst:[fd9c:1c81:262e:1:67f6:b2f6:c04b:2a48]:33521
```

aaa147147, 2024-12-03:
@abtink Thank you for your reply. 
To improve troubleshooting when full logs aren't available, could we add a reason to each "error: Drop" message?
 For example, "error: AddressQuery". This would provide more context without needing full logs.

abtink, 2024-12-03:
> To improve troubleshooting when full logs aren't available, could we add a reason to each "error: Drop" message?

The reason is already included in the log as `error:Drop`, which indicates device has determined there is no route to this destination (which the device can reach due to a variety of earlier events or failures such as an earlier address query timeout)

> For example, "error: AddressQuery". This would provide more context without needing full logs.

This would not be proper/correct, as there is no "Address Query" attempt made for this message.

aaa147147, 2024-12-04:
@abtink Okay, I understand the meaning. When Drop occurs, it means that the destination address is unreachable. Thank you for your reply

closed 2024-12-18 by aaa147147