openthread/openthread #11063: Teardown of IPv6 routes on loss of radio

agners, 2024-12-17:
**Describe the bug** A clear and concise description of what the bug is.

When the radio is no longer reachable (e.g. if the host USB bus has issues or the radio is otherwise connected through an unreliable connection) the OTBR might shutdown due to transmit errors such as `radio tx timeout`:

```
Dec 16 22:56:00 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:34.781 [W] P-HdlcIntface-: Error decoding hdlc frame: Parse
Dec 16 22:56:00 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:34.819 [W] P-HdlcIntface-: Error decoding hdlc frame: Parse
Dec 16 22:56:00 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:34.822 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:96, src:0x2000, dst:0xcc00, sec:yes, tag:61645, offset:592, dglen:780
Dec 16 22:56:00 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:34.849 [N] MeshForwarder-: Dropping rx frag frame, error:Drop, len:92, src:0x2000, dst:0xcc00, sec:yes, tag:61645, offset:688, dglen:780
Dec 16 22:56:01 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:35.562 [N] MeshForwarder-: Dropping (reassembly queue) IPv6 UDP msg, len:780, chksum:b3c9, ecn:no, sec:yes, error:ReassemblyTimeout, prio:normal, rss:-65.0, radio:15.4
Dec 16 22:56:01 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:35.562 [N] MeshForwarder-:     src:[fd25:bb06:3d86:1:537d:b4c4:a327:13a2]:5540
Dec 16 22:56:01 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:35.562 [N] MeshForwarder-:     dst:[fd3a:ce91:4b25:0:8c0e:67ca:52c4:87a4]:45556 
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [W] P-RadioSpinel-: radio tx timeout
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [C] P-RadioSpinel-: Failed to communicate with RCP - no response from RCP during initialization
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [C] P-RadioSpinel-: This is not a bug and typically due a config error (wrong URL parameters) or bad RCP image:
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [C] P-RadioSpinel-: - Make sure RCP is running the correct firmware
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [C] P-RadioSpinel-: - Double check the config parameters passed as `RadioURL` input
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: 00:15:37.552 [C] Platform------: HandleRcpTimeout() at radio_spinel.cpp:2011: RadioSpinelNoResponse
Dec 16 22:56:03 hass-otbr-boven addon_core_openthread_border_router[445]: [23:56:03] WARNING: otbr-agent exited with code 6 (by signal 0).
```

In that case, the OTBR does not tear down the RoutingManager, hence leaving the routes on other hosts as valid.

This is different to a OTBR which gets disconnected from the network entirly: Typically reachability probe for the router fail in these cases, so the other systems will automatically choose an alternative route.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id cc8f078
2. IEEE 802.15.4 hardware platform EFR32MG21
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

Ideally the routing maanger would send out an updated RA with a lifetime 0 like on a regular shutdown:
```
00:00:09.975 [I] RoutingManager: Disabling
00:00:09.975 [I] RoutingManager: Stopping Nat64PrefixManager
00:00:09.975 [I] RoutingManager: Preparing RA
00:00:09.975 [I] RoutingManager: - RA Header - flags - M:1 O:1 S:1
00:00:09.975 [I] RoutingManager: - RA Header - default route - lifetime:0
00:00:09.975 [I] NetDataPublshr: Unpublishing ::/0
00:00:09.975 [I] NetDataPublshr: ExternalRoute ::/0 - State: Added -> NoEntry
00:00:09.975 [I] RoutingManager: Stopped
```

**Console/log output** If applicable, add console/log output to help explain your problem.

**Additional context** Add any other context about the problem here.

superwhd, 2024-12-18:
That's a good point. Probably we can add some cleanup steps when the program exits. 

Not sure what mechanism should be used though: RAII, signal, std::set_terminate, etc.

abtink, 2024-12-18:
The `RoutingManager` is automatically stopped whenever the device enters the "disabled/detached" role.

The notion of "loss of radio" is not well-defined, at least in the radio platform abstraction (`otPlatRadio` APIs). From the perspective of the OT core, once the radio is enabled, it is expected to function.

I suggest using a platform-specific solution to determine that the radio is no longer available. For example, `RadioSpinel` can determine this by its inability to communicate with RCP. This solution can then invoke `otThreadSetEnabled()` to stop Thread/MLE operation, which should trigger a role change to "disabled" and stop the `RoutingManager`.

jwhui, 2024-12-18:
I agree with @abtink here.

We should update radio spinel to call `otThreadDetachGracefully()` to stop Thread operation, which will in turn clean up everything else like routing manager, border agent, etc. Then exit the process within the associated callback.

agners, 2025-01-11:
Hm, I see `otThreadDetachGracefully()` is asynchronous operation. I wonder if that can work if the spinel link is down? :thinking: 

FWIW, we currently enabled `OPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT` to deal with RCP instability issues, in that case the spinel protocol stays in operation by default.

However, even with `OPENTHREAD_SPINEL_CONFIG_RCP_RESTORATION_MAX_COUNT` enabled, there could be the situation where the RCP just doesn't respond (leading to `Too many rcp failures, exiting`). In that case, a graceful detach would still be better.

jwhui, 2025-01-14:
I expect `otThreadDetachGracefully()` to still perform appropriate cleanup even if the spinel link is down. Of course, it will not be able to send messages over the radio, but it should still be able to perform any local cleanup.