openthread/openthread #11103: Bricked device because of premature removal of keys from datasets on moving them to secure storage

hubertmis, 2025-01-01:
**Describe the bug**

The dataset manager includes a method `MoveKeysToSecureStorage` that seems to perform the following steps:

1. For each key in the dataset
    1. Copy key value from the dataset to the secure storage
    2. Replace the key value in the dataset with zeros
2. Save the cleared dataset in the non-volatile storage

When this procedure is applied, the firmware is not capable anymore to use the keys from the dataset, because they are replaced with zeros. However, this method does not account for emergency rollback procedure on a firmware update failure.

When the emergency rollback is applied, it is possible that:

1. New firmware image moves keys to the secure storage and removes them from the dataset
2. Because of any reason, the new firmware is considered invalid
3. A rollback is performed to restore the previous valid firmware
4. The previous firmware uses keys from the dataset (not from the secure storage). It cannot connect to the network anymore, because the "new firmware" already cleared the keys on its initialization.

**To Reproduce**

I encountered this issue with nRF52840 upgrading nRF Connect SDK from v1.8.0 (openthread commit c02d111026d05c841354b9e36d1f1e2e9ac929c2) to v2.8.0 (openthread commit 2aeb8b833ba760ec29d5f340dd1ce7bcb61c5d56 with one cherry-pick: 4c45ee8e0e550e4ff78e0799f86d9a38bfadbd28).

**Expected behavior**

I expect the emergency rollback on failed update to be possible.

I think the function `MoveKeysToSecureStorage` should copy the keys to the secure storage on the protocol initialization. However, it should restrain from clearing key values from the dataset until the firmware update is confirmed to be successful.

abtink, 2025-01-01:
@hubertmis, I think this behavior is expected and normal. 

Enabling `OPENTHREAD_CONFIG_PLATFORM_KEY_REFERENCES_ENABLE` (Secure Storage and ARM PSA) significantly changes how security keys are stored.

The OT stack manages the transition from firmware without Secure Storage to one that uses it by moving keys from the dataset's `Settings` to PSA Secure Storage. However, reverting to firmware without `KEY_REFERENCES_ENABLE` will not work because the code for reading persisted keys from PSA secure storage is absent in the old firmware.

This is analogous to changing the non-volatile `otPlatSettings` platform implementation (e.g., using a new settings format). A common solution for enabling rollback to older firmware with different `Settings` formats is to use a (vendor-specific) transitional firmware that reads the persisted data using the new format and saves it using the old format before rollback.

> I think the function `MoveKeysToSecureStorage` should copy the keys to the secure storage on protocol initialization. However, it should restrain from clearing key values from the dataset until the firmware update is confirmed to be successful.

Storing keys in both secure storage and the dataset defeats the purpose of using secure storage, which is intended to provide enhanced security compared to regular settings.

It is not clear how the OT stack would define or determine "until the firmware update is confirmed to be successful." The OT stack generally assumes it is functional.

hubertmis, 2025-01-02:
> It is not clear how the OT stack would define or determine "until the firmware update is confirmed to be successful."

I was thinking about a function called by the firmware updating module at the same time the new firmware is marked as functional (after post-update tests pass).

> This is analogous to changing the non-volatile `otPlatSettings` platform implementation (e.g., using a new settings format).

The main difference I see is that `MoveKeysToSecureStorage` is done silently by OpenThread without any indication of the potential risk. If I replaced `otPlatSettings` implementation it would be manual action on my side and I would be aware of the risks. And I would need to keep both old and new settings format temporarily to allow for the emergency rollback. My settings-structure-update algorithm would remove (and free memory containing) old settings after the update is marked as functional and I would be sure the emergency rollback won't happen anymore.

Or maybe `OPENTHREAD_CONFIG_PLATFORM_KEY_REFERENCES_ENABLE` already documents the risk associated with enabling this option but I simply missed the documentation?

abtink, 2025-01-02:
Here are my suggestions:
- Regarding document updates, it may help to add more explicit warnings to ensure other platform vendors understand the impact of enabling `KEY_REFERENCES_ENABLE` and what it would entail.
    - "Secure Storage" (ARM-PSA) itself was contributed by interested platform vendors (in #6862).
- It may help to add a new API/mechanism to allow the run-time enabling/disabling of "Secure Storage" (ARM PSA) behavior.
    - This would then update any already saved/persisted information.
    - I expect this would be a bigger change in the code, so it would be good to accompany it with tests to validate it.
    - There were suggestions to add simulation tests for "secure storage" in the original PR -> [https://github.com/openthread/openthread/pull/6862#issuecomment-911971884](https://github.com/openthread/openthread/pull/6862#issuecomment-911971884) (which I don't think is included).

---

> The main difference I see is that `MoveKeysToSecureStorage` is done silently by OpenThread without any indication of the potential risk. If I replaced the `otPlatSettings` implementation, it would be a manual action on my side, and I would be aware of the risks.

Enabling `OPENTHREAD_CONFIG_PLATFORM_KEY_REFERENCES_ENABLE` also requires action by the platform vendor to implement and provide a whole new set of `otPlatCrypto` APIs related to key-ref support, such as `otPlatCryptoImportKey()`, `otPlatCryptoExportKey()`, etc.

I assume the original authors of "Secure Storage" (ARM-PSA) expected its impact and risks to be clear.

> I was thinking about a function called by the firmware updating module at the same time the new firmware is marked as functional (after post-update tests pass).

Adding an API in OT to say, "firmware is now functional" seems rather weird to me. I would expect a platform/product integrator to validate and ensure its complete firmware is functional and passes all tests before pushing it to devices through a software update. So from the OT stack perspective, it should assume it is always functional and error-free.

hubertmis, 2025-01-03:
> Enabling `OPENTHREAD_CONFIG_PLATFORM_KEY_REFERENCES_ENABLE` also requires action by the platform vendor to implement and provide a whole new set of `otPlatCrypto` APIs related to key-ref support, such as `otPlatCryptoImportKey()`

I see. I missed this because it was added by SDK implementation for me.

> I would expect a platform/product integrator to validate and ensure its complete firmware is functional and passes all tests before pushing it to devices through a software update.

Of course, the firmware passes all the tests on the test bench before it is pushed to the device. However, not all configuration settings combinations can be tested in a test bench and some problems are discovered only in the deployed devices after the firmware is pushed. If problems are discovered there, an emergency procedure gets activated. It might be entering an emergency update port (Bluetooth, UART, etc) or, more user-friendly, emergency roll-back to the previous functional image version. I would like OpenThread to work correctly with the emergency roll-back procedure.

I have one more idea on how to make it functional:

How about removing `MoveKeysToSecureStorage` from the automatic protocol initialization sequence and requiring the firmware updating procedure to call two functions instead:

1. `CopyKeysToSecureStorage` - while updating, before OpenThread is initialized
2. `PurgeKeysFromDataset` - after the firmware confirms the emergency rollback won't be applied

abtink, 2025-01-03:
> I would like OpenThread to work correctly with the emergency roll-back procedure.

While this would be useful, I think this needs to be vendor/product-specific. Any backward-incompatible change to non-volatile settings will need to be handled specifically. Enabling PSA "Secure Storage" is one example of this.

OT can provide helper APIs to facilitate this, but providing a general-purpose solution for all such situations does not seem possible (or can get very complex).

---

> How about removing `MoveKeysToSecureStorage` from the automatic protocol initialization sequence and requiring the firmware updating procedure to call two functions instead:
>
> 1. [...]

I am not in favor of this:

- It requires the app/next layer to be involved and call these new APIs.
    - This can be easily missed or forgotten by platform/app developers.
    - I expect such a change will lead to folks asking why it is not done automatically. :)
- It only allows the immediate roll-back use-case (after PSA "secure storage" is enabled for the first time), but it cannot handle rolling back after `PurgeKeysFromDataset` is performed.

----

My suggestion is still to add a new API/mechanism to allow the run-time enabling/disabling of "Secure Storage" (ARM PSA) behavior. When the API is called to disable, the code should read any saved information in "secure storage" and save it again in regular `Settings` before deleting the information in "secure storage."

This can then be used to update firmware to versions that don't support the PSA storage feature (whether rolling back to older firmware or using new firmware without this feature).

The software update (or roll-back) mechanism would still be vendor/product-specific:
-   It can use intermediate, transitional firmware that is pushed to perform this settings update before main sw update.
-   Or this can be part of platform initialization and included in every firmware.
-   Etc.

jwhui, 2026-11-13:
Closing per the discussion above. Enabling `OPENTHREAD_CONFIG_PLATFORM_KEY_REFERENCES_ENABLE` intentionally migrates dataset keys out of plaintext `otPlatSettings` into secure storage as a one-way storage transition, and managing OTA rollback across storage-format changes is specific to the platform/product update design. Feel free to open a PR or discussion if you would like to propose optional migration helper APIs.

closed 2026-11-13 by jwhui