openthread/openthread #11233: AMD64 Docker image not pushed to latest tag.

thenoid, 2025-02-10:
**Describe the bug** A clear and concise description of what the bug is.

Not sure if this is the right github project, sorry if it isn't.  The `latest` docker image was pushed without a `amd64` image.

https://hub.docker.com/r/openthread/otbr/tags

This results in a `no matching manifest for linux/amd64 in the manifest list entries` error

**To Reproduce** Information to reproduce the behavior, including:

```root@donnager:~# docker pull openthread/otbr
Using default tag: latest
latest: Pulling from openthread/otbr
no matching manifest for linux/amd64 in the manifest list entries```

jwhui, 2025-02-10:
Thanks for raising this issue.

I believe this is due to https://github.com/openthread/ot-br-posix/pull/2698, where the build for different platforms was split out into different runners. What's missing is the step to combine them in a single manifest.

I'll try to look into this when I find some time. But if others are willing to contribute, feel free to submit a PR!

closed 2025-02-12 by jwhui

thenoid, 2025-02-12:
Thx Sir!