openthread/openthread #11255: Device Reset (Bus Fault) Under High-Frequency TCP Connection Stress

dsyx, 2025-02-14:
**Describe the bug**

I'm developing a TCP-based server that only allows a single connection at a time.  When a new connection request arrives, the server calls `otTcpAbort` on the existing connection within the Accept Ready callback and then accepts the new request.

Under stress testing, where TCP connections are rapidly established, data is sent/received, and then disconnected, the device eventually resets with a Bus Fault.

Here's the relevant part of the reset log:
```
[C] Platform------: Reset info: 0xa (FLT)
[C] Platform------: Extended Reset info: 0xA03 (BUS)
[C] Platform------: Thread mode using process stack (   0 to    0), SP = 200040f0
[C] Platform------: 8 bytes used (0%) in process stack (out of 0 bytes total)
[C] Platform------: SP is outside process stack range!
[C] Platform------: No interrupts active
[C] Platform------: Reset cause: Bus Fault
[C] Platform------: Instruction address: 807d108
[C] Platform------: Illegal access address:    8
[C] Platform------: CFSR.PRECISERR: precise data bus fault
[C] Platform------: CFSR.BFARVALID: BFAR contains valid fault address
[C] Platform------: CFSR.(none) load or store at an illegal address
[C] Platform------: R0 = 0x0000000f, R1 = 0x00000000, R2 = 0x0000000f, R3 = 0x200041bc
[C] Platform------: R4 = 0x00000000, R5 = 0x00000019, R6 = 0x00000010, R7 = 0x0000000c
[C] Platform------: R8 = 0x00000020, R9 = 0x0000000a, R10 = 0x00000000, R11 = 0x00000905
[C] Platform------: R12 = 0x00000000, R13(LR) = 0xfffffffd, MSP = 0x200011a0, PSP = 0x200040f0
[C] Platform------: PC = 0x0807d108, xPSR = 0x61100200, MSP used = 0x00000630, PSP used = 0x00000008
[C] Platform------: CSTACK bottom = 0x20000008, ICSR = 0x00000805, SHCSR = 0x000f0002, INT_ACTIVE0 = 0x00000000
[C] Platform------: INT_ACTIVE1 = 0x00000000, CFSR = 0x00000000, HFSR = 0x00008200, DFSR = 0x00000000
[C] Platform------: MMAR/BFAR = 0x00000000, AFSR = 0x00000008, Ret0 = 0x00000000, Ret1 = 0x08006f55
[C] Platform------: Ret2 = 0x00000000, Ret3 = 0x00000000, Ret4 = 0x00000000, Ret5 = 0x00000000
[C] Platform------: Dat0 = 0x00000000, Dat1 = 0x0df7a8e2, 
```

Debugging suggests the issue occurs within the `lbuf_getrange` function.  Specifically, a breakpoint revealed that the `buffer->head` pointer passed to `lbuf_getrange` is `NULL`, leading to the illegal memory access.

[image]

Should `lbuf_getrange` include parameter validation to check for a NULL `buffer->head` pointer?

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id `7f6723ffb`, that is the version referenced by [Simplicity SDK 2024.12.0](https://github.com/SiliconLabs/simplicity_sdk/commit/8627f8482564dc6d94b56512740a39d6f409a0eb)
2. IEEE 802.15.4 hardware platform: `efr32`

romacdon, 2025-02-14:
@dsyx Thanks for reporting the issue.  Can you apply the change in this PR and see if it resolves the issue? https://github.com/openthread/openthread/pull/10977

dsyx, 2025-02-14:
Hi @suveshpratapa ,

I used a shell script to test the device. I don't have the script at hand now, but it works roughly like this (loop):

1. Generate 1 - 10 random messages, each containing 1 - 256 characters.
2. Connect to the device and send messages with an interval of 100ms between messages.
3. Send a message to the device, and the device will echo the message back to the sender when it receives the message.
4. Disconnect and sleep for 1 - 5 seconds.

I used this shell script to start two program instances to test the scenario of "disconnecting the current connection and accepting a new connection as soon as a new connection request is made".

On the device, I used `otTcpAbort` in the Accept Ready callback to disconnect the current connection. I used the silabs ot-ble-dmp example and used `otTcpSendByReference` in the ot_app_task to send the message.

dsyx, 2025-02-17:
I tried `otTcpCircularSendBufferWrite` instead of `otTcpSendByReference`. As a result, the device is stuck on `OT_ASSERT(aInSendBuffer <= mCapacityUsed);`, where `aInSendBuffer` is `15` and `mCapacityUsed` is 0. It seems that `otTcpAbort` cannot be called in the Accept Ready callback.

dsyx, 2025-02-17:
In the case of the circular send buffer, commenting out the `otTcpCircularSendBufferForceDiscardAll` call in the Disconnected callback, the device no longer gets stuck and appears to be working properly.

dsyx, 2025-02-18:
Hi @suveshpratapa ,

I just used CLI to simulate my usage scenario.
 
- TCP establishes a connection, receives data, and then aborts it. This is OK.
- TCP establishes a connection, receives data, sends data, and then aborts it. It will assert.

```
> tcp init
Done
> tcp listen :: 30000
Done
> Accepted connection from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55244
TCP: Connection established
TCP: Received 190 bytes: kYg56gFdhNA5FNWqZxl0iKWLTetQLu8suZahuOjvLi06U6vKfLc4wYQSXCgvmr4JQVJ2/2rs9fC3ZD8lwefa+IcivH3vp361DADUPXoatA7uL+Gx+Z9u34H5FonazWtkLfMORhMeMkNtaHI7qZ3FHmfwsfcmW3K9aBLzrmZnCrQOQAliH+NP+BfNZqX20

TCP: Reached end of stream
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252 (active socket is busy)
tcp abort
TCP: Connection reset
Done
> Accepted connection from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:55252
TCP: Connection established
TCP: Received 68 bytes: FJsNKB+q1dv0gH2vTwt91axmim+EzBGBGIeSuCiF8gRIO0vpa6pmaU+exnoOyKJLwn7

TCP: Reached end of stream
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
TCP: Ignoring incoming connection request from [fd6f:4db:fbd:1:66c4:1ee:7728:185a]:39098 (active socket is busy)
tcp send helloworld
Done
> TCP: Connection reset
tcp abort
TCP: Connection reset
Done
> Accepted connection from 
```

dsyx, 2025-02-18:
Hi @samkumar ,

It looks like `otTcpAbort` doesn't handle the send buffer well. I added the following code and the CLI works:

```diff
void tcplp_sys_on_state_change(struct tcpcb *aTcb, int aNewState)
{
    if (aNewState == TCP6S_CLOSED)
    {
        /* Re-initialize the TCB. */
+       uint32_t ntraversed;
+       lbuf_pop(&aTcb->sendbuf, lbuf_used_space(&aTcb->sendbuf), &ntraversed);
        cbuf_pop(&aTcb->recvbuf, cbuf_used_space(&aTcb->recvbuf));
        aTcb->accepted_from = nullptr;
        initialize_tcb(aTcb);
    }
    /* Any adaptive changes to the sleep interval would go here. */
}
```

If there is a problem with my changes, please let me know.

dsyx, 2025-04-23:
Hi @suveshpratapa ,

Pull request #11269 has been merged into main, please consider syncing to the Simplicity SDK.

closed 2025-04-23 by dsyx