openthread/openthread #11261: Follow up Issue: Address query resolution fails when there is link quality mismatch (no incoming link)

sarveshkumarv3, 2025-02-18:
**Describe the bug** A clear and concise description of what the bug is.

This is follow up to a previous issue raised and fixed via https://github.com/openthread/openthread/issues/9949

In this case, under fringe conditions, the incoming link quality to neighbor is 0 but there is an outgoing link.  In such a scenario if address query procedure is triggered, it causes Address Notify ACK to be dropped due to NoRoute till there is an advertisement with Route TLV to update the incoming link quality. 

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.

Expected link to be reestablished to resume communication between 2 devices when Address query resolution reaches AddressNotify stage. 

**Console/log output** If applicable, add console/log output to help explain your problem.

//Incoming link quality is 0 to node 0x4c00 

08:24:47.464 [I] RouterTable---: Route table
08:24:47.464 [I] RouterTable---:     13 0x3400 - me - leader
08:24:47.465 [I] RouterTable---:     47 0xbc00 - nbr{lq[i/o]:0/1 cost:16} - nexthop{37 cost:1}
08:24:47.465 [I] RouterTable---:     37 0x9400 - nbr{lq[i/o]:2/2 cost:2} - nexthop{19 cost:1}
08:24:47.465 [I] RouterTable---:     19 0x4c00 - nbr{lq[i/o]:0/2 cost:16} 
08:24:47.465 [I] RouterTable---:     30 0x7800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
08:24:47.465 [I] RouterTable---:     33 0x8400 - nbr{lq[i/o]:3/3 cost:1} - nexthop{30 cost:1}
08:24:47.465 [I] RouterTable---:     42 0xa800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{33 cost:1}
08:24:47.465 [I] RouterTable---: Route table dump end

//Address query sent to resolve an address belonging to 0x4c00 

08:25:12.649 [I] AddrResolver--: Sent AddressQuery for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e
08:25:12.649 [I] AddrResolver--: Cache entry added: fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e, 0xfffe - query request
08:25:12.670 [I] AddrResolver--: Received AddressNotify from 0x4c00 for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e to 0x4c00
08:25:12.670 [I] AddrResolver--: Cache entry updated: fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e, 0x4c00 - rx notification
08:25:12.670 [I] AddrResolver--: Sent AddressNotify ack
08:25:12.670 [W] Mle-----------: Check reachability mesh dst: 4c00, IPv6 src: [fdb6:5ff6:908d:705:0:ff:fe00:3400], ipv6 dst: [fdb6:5ff6:908d:705:0:ff:fe00:4c00], not able to generate next hop

This behavior continues till Advertisement is received from 0x4c00

abtink, 2025-02-25:
@sarveshkumarv3 , If I am not mistaken, I think this is already implemented and covered by the following PR.

- https://github.com/openthread/openthread/pull/10575

> This commit introduces a freshness timeout mechanism for address cache entries which is used to decide when removing stale entries when the associated RLOC16 is unreachable.
> 
> In `AddressResolver`, when resolving an EID from an existing cache entry, if the target RLOC16 is unreachable, the entry is typically considered stale and removed to allow a new address query to be sent. This commit adds a mechanism to skip this removal step if the entry has been recently updated, i.e., an `AddressNotify` has been received for it and its `FreshnessTimeout` has not yet expired.
> 
> The `FreshnessTimeout` check prevents repeated address query transmissions when mesh routes are not yet discovered (e.g., after initial attach) or if there is a temporary link issue.

I also found this related issue https://github.com/openthread/openthread/issues/9949 (which is the origin of [SPEC-1308](https://threadgroup.atlassian.net/browse/SPEC-1308) and its related PR  https://github.com/openthread/openthread/pull/10906 and also covers the above PR about freshness).

sarveshkumarv3, 2025-02-27:
thanks @abtink .@jwhui I do not know the exact commit ID used on the accessory side (which is a third-party commercial accessory), but on the router side, the base git commit ID is older 
 ( d9abe3071c0131a4adb5d7e7451319b735e6d855)  but it has the below PRs cherry-picked

https://github.com/openthread/openthread/pull/10906
https://github.com/openthread/openthread/pull/10575

The issue was seen in targeted testing in far field scenarios where RSSI was poor but still there was a link.  https://github.com/openthread/openthread/pull/10575 avoids the address query storm and limits the address query to every 3s, however the packet drops in this scenario is still seen, as the router drops due to NoRoute. https://github.com/openthread/openthread/pull/10906 addresses the link mismatch when outgoing link quality is 0, but the issue seen here is the opposite, where the incoming link quality is 0 but the outgoing link quality is 2. 

Please find the pcap log attached. Time of issue is between 14:59:20 and 14:59:32

Key=1F15BF8E1E5AF41323014FC1DD75B1EB
KeyIndex=2

[wireshark_capture.pcap.zip](https://github.com/user-attachments/files/19005109/wireshark_capture.pcap.zip)

abtink, 2025-02-27:
@sarveshkumarv3, it seems the accessory side does not include the #10906 fix.

If a device concludes that it has no link to a specific neighbor (often due to missing ACKs on many transmission retries), it marks the neighbor as unreachable (marking its "Link Quality Out" as zero).

- If there is another path (through another router), communication can proceed using multiple hops. But if there is no other path, the `NoRoute` error is the correct behavior, and messages will be dropped.
- This will be resolved when the device receives and processes an advertisement from the neighbor to update its route table.
- The enhancement in #10906 addresses this situation by having the neighbor device detect this and send an advertisement more quickly.

**If you see that the situation is resolved once the advertisement is received, then you are observing the same issue.**

Regarding this, we have already implemented enhancements (the above PRs) to help with this specific situation. Do you have any suggestions/ideas for anything else to be done for this? 

---

> [https://github.com/openthread/openthread/pull/10906](https://github.com/openthread/openthread/pull/10906) addresses the link mismatch when the outgoing link quality is 0, but the issue seen here is the opposite, where the incoming link quality is 0 but the outgoing link quality is 2.

This is very different from earlier topics.

- "Link Quality In" is determined locally on the device itself (from the RSS of received messages from the neighbor).
- "Link Quality Out" is learned from processing received `RouteTlv` in the advertisements sent by the neighbor (it is the link quality the neighbor claims to see when it receives from us, its "link quality in" to us).

If we have a nonzero "Link Quality Out," it means we have received an advertisement from the neighbor, and therefore we will generally have a nonzero "Link Quality In" as well (we have at least one RSS sample).

If "Link Quality In" is 0, it indicates that the RSS of the received messages from the neighbor is so low that the link is deemed unusable. In this case, it does not matter that the neighbor claims to be able to receive well from us; the bidirectional link quality is poor, and not using this link to communicate with the neighbor is the correct behavior.

I want to make sure that you are correct here about "opposite". Are you maybe observing "Link Quality Out" being zero and "Link Quality In" being 2 (toward a neighbor) on the original device versus seeing the same from the perspective of the neighbor?

sarveshkumarv3, 2025-02-27:
Thanks @abtink 

>I want to make sure that you are correct here about "opposite". Are you maybe observing "Link Quality Out" being zero and "Link Quality In" being 2 (toward a neighbor) on the original device versus seeing the same from the perspective of the neighbor?

I am observing link quality In being 0 and link quality Out being 2 on the original devices towards the neighbor 

> If "Link Quality In" is 0, it indicates that the RSS of the received messages from the neighbor is so low that the link is deemed unusable. In this case, it does not matter that the neighbor claims [...]

>Regarding this, we have already implemented enhancements (the above PRs) to help with this specific situation. Do you have any suggestions/ideas for anything else to be done for this?

I think in this scenario, the original device does not have a route via any next hop. It has an outgoing link with non-zero link quality (2) but we are not using it because the incoming RSSI is too low (link quality in is 0). 

One optimization could be,  to not use two-way link quality based computation in this scenario (when there is no route via  next hop)  and just use the outgoing link quality (as a weak link still exists).

abtink, 2025-02-27:
You mentioned in the first comment:

> This behavior of address query storm continues till Advertisement is received to reestablish incoming link quality for 0x4c00

I would expect receiving an advertisement from the neighbor should not immediately address the issue (unless the RSS of received message from the neighbor suddenly improves significantly). Can you please confirm this is the case?

The "Link Quality In" is calculated from the current link margin, itself determined from a running average RSS over received frames from the neighbor. A hysteresis is also applied:

https://github.com/openthread/openthread/blob/b5164cf836d28b83f9123ec04c0e4189e0da1723/src/core/thread/link_quality.hpp#L342-L355

So I expect the change to be gradual and receiving one (or may be even few) new frames (say advertisements) from neighbor should not immediately impact this.

Understanding why or how the device ended up setting "link quality in" to zero for a neighbor may also be helpful. In case you have the logs and can see what/how it happened.

sarveshkumarv3, 2025-02-28:
@abtink please find the sequence below. 

>I would expect receiving an advertisement from the neighbor should not immediately address the issue (unless the RSS of received message from the neighbor suddenly improves significantly). Can you please confirm this is the case?

Sorry one correction regarding my previous observation. The address query storm continues but eventually there is no AddressNotify response coming back and hence times out. It does not recover on receiving Advertisement. 

>So I expect the change to be gradual and receiving one (or may be even few) new frames (say advertisements) from neighbor should not immediately impact this.

Yes, the incoming link quality reaches 0 gradually. 

03:32:44.999 [I] RouterTable---: Route table
03:32:44.999 [I] RouterTable---:     13 0x3400 - me - leader
03:32:44.999 [I] RouterTable---:     30 0x7800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
03:32:44.999 [I] RouterTable---:     33 0x8400 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
03:32:44.999 [I] RouterTable---:     42 0xa800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{30 cost:1}
03:32:44.999 [I] RouterTable---:     19 0x4c00 - nbr{lq[i/o]:1/2 cost:4} - nexthop{47 cost:1}
03:32:44.999 [I] RouterTable---:     37 0x9400 - nbr{lq[i/o]:1/2 cost:4} - nexthop{19 cost:1}
03:32:44.999 [I] RouterTable---:     47 0xbc00 - nbr{lq[i/o]:1/2 cost:4} - nexthop{19 cost:1}
03:32:44.999 [I] RouterTable---: Route table dump end

The next frames received from 0x4c00 are with the below RSSI 

03:32:48.409 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -95
03:33:49.093 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -96
03:34:19.842 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00)  - RSSI: -96
03:35:00.133 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00)  - RSSI:  -96
03:35:31.074 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00)  - RSSI: -98
03:36:03.750 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00)  - RSSI: -95
03:36:27.884 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -96
03:37:06.469 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -96
03:37:29.077 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -97
03:38:01.306 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -97 
 03:38:37.758 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -97 
03:39:06.754 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -98
03:39:38.842 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -98
03:40:18.372 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -100
03:40:45.992 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -99
03:41:53.638 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -100
03:42:19.036 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -99

03:42:24.376 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:9ce6:209d:597f:8943,0x9400)

 03:42:24.376 [I] RouterTable---: Route table
 03:42:24.376 [I] RouterTable---:     13 0x3400 - me - leader
 03:42:24.376 [I] RouterTable---:     30 0x7800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
 03:42:24.376 [I] RouterTable---:     33 0x8400 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
 03:42:24.376 [I] RouterTable---:     42 0xa800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{30 cost:1}
 03:42:24.376 [I] RouterTable---:     19 0x4c00 - nbr{lq[i/o]:0/2 cost:16} - nexthop{47 cost:1}
 03:42:24.376 [I] RouterTable---:     37 0x9400 - nbr{lq[i/o]:1/2 cost:4} - nexthop{19 cost:1}
 03:42:24.376 [I] RouterTable---:     47 0xbc00 - nbr{lq[i/o]:2/2 cost:2} - nexthop{37 cost:1}
 03:42:24.376 [I] RouterTable---: Route table dump end

03:42:57.792 [I] Mle-----------: Receive Advertisement (fe80:0:0:0:cc8:fec4:400a:7dcc,0x4c00) - RSSI: -100

03:42:58.349 [I] AddrResolver--: Sent AddressQuery for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e
03:42:58.349 [I] AddrResolver--: Cache entry added: fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e, 0xfffe - query request
03:42:58.374 [I] AddrResolver--: Received AddressNotify from 0x4c00 for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e to 0x4c00
03:42:58.374 [I] AddrResolver--: Cache entry updated: fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e, 0x4c00 - rx notification
03:42:58.374 [I] AddrResolver--: Sent AddressNotify ack
03:42:58.374 [W] MeshForwarder-: Dropping IPv6 UDP msg, src=[fdb6:5ff6:908d:705:0:ff:fe00:3400]:61631, dst=[fdb6:5ff6:908d:705:0:ff:fe00:4c00]:61631, len=52, chksum=f6ab, ecn=no, sec=yes, error=NoRoute

03:43:01.361 [I] AddrResolver--: Sent AddressQuery for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e
03:43:01.386 [I] AddrResolver--: Received AddressNotify from 0x4c00 for fd82:416:f0bb:0:87e3:d6e7:37c3:2f3e to 0x4c00
03:43:01.386 [I] AddrResolver--: Sent AddressNotify ack
03:43:01.386 [I] MeshForwarder-: Dropping IPv6 UDP msg, src=[fdb6:5ff6:908d:705:0:ff:fe00:3400]:61631, dst=[fdb6:5ff6:908d:705:0:ff:fe00:4c00]:61631, len=52, chksum=f6aa, ecn=no, sec=yes, error=NoRoute

Above behavior for address query continues 

[... cut ...]
Till the router ID is released due to NoRoute around the below time 

03:53:30.043 [N] RouterTable---: Release router id 19
03:53:30.043 [I] RouterTable---: Route table
03:53:30.043 [I] RouterTable---:     13 0x3400 - me - leader
03:53:30.043 [I] RouterTable---:     30 0x7800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
03:53:30.043 [I] RouterTable---:     33 0x8400 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
03:53:30.043 [I] RouterTable---:     42 0xa800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{30 cost:1}
03:53:30.043 [I] RouterTable---: Route table dump end

abtink, 2025-02-28:
Thanks. A quick, unrelated topic: I immediately noticed the line `[I] RouterTable---: Route table dump end`. I am pretty sure this log is not in OT code (did double-check it to make sure I am not losing my memory 😄). Similarly, this line looks different. OT uses `:` and not `=` as in `len:52` vs. `len=52`.
```
MeshForwarder-: Dropping IPv6 UDP msg, src=[fdb6:5ff6:908d:705:0:ff:fe00:3400]:61631, dst=[fdb6:5ff6:908d:705:0:ff:fe00:4c00]:61631, len=52, chksum=f6aa, ecn=no, sec=yes, error=NoRoute
```

I guess I have looked at enough OT logs over the years (and actually have written much of the logging code), so I can tell when something looks strange or different. 😃 

Just curious, is there any reason for these changes to the logs?

sarveshkumarv3, 2025-02-28:
Thanks. There are some logging format changes to match some internal post-processing scripts. These are part of those changes.

abtink, 2025-02-28:
Regarding the link to `0x4c00`, the RSSI is indeed very low, and the link quality is poor, so a "Link Quality In" of zero is correct, and not using the direct link to communicate with this device (effectively setting the cost of the direct link to 16) is also correct. This results in dropping messages, which is fine (I believe it is the desired behavior per the spec).

Regarding Address Query failures, I expect #10575 to apply rate-limiting and limit the number of attempts (I think the freshness timeout was 3 seconds?). Can you please check and confirm that this is what you observe?

I know you are suggesting that maybe we allow the node to still use the direct link (with "link quality zero") to communicate with `0x4c00` (if there is no other option). This may be helpful in this particular test/setup, but I am concerned about the impact of such a change in general. This would be effectively the same as updating the definition of "link quality 0" (using a lower threshold for its definition). I feel somewhat uncomfortable with such a change, and my gut feeling says that it may unintentionally break a lot of other behaviors.

If the "link quality in" is zero, then we should not use the direct link, but if we find a different (indirect) path to reach this device through multi-hop, we can use it instead. For example, we see at some point in the logs:

```
 03:42:24.376 [I] RouterTable---: Route table
 03:42:24.376 [I] RouterTable---:     13 0x3400 - me - leader
 03:42:24.376 [I] RouterTable---:     30 0x7800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
 03:42:24.376 [I] RouterTable---:     33 0x8400 - nbr{lq[i/o]:3/3 cost:1} - nexthop{42 cost:1}
 03:42:24.376 [I] RouterTable---:     42 0xa800 - nbr{lq[i/o]:3/3 cost:1} - nexthop{30 cost:1}
 03:42:24.376 [I] RouterTable---:     19 0x4c00 - nbr{lq[i/o]:0/2 cost:16} - nexthop{47 cost:1}
 03:42:24.376 [I] RouterTable---:     37 0x9400 - nbr{lq[i/o]:1/2 cost:4} - nexthop{19 cost:1}
 03:42:24.376 [I] RouterTable---:     47 0xbc00 - nbr{lq[i/o]:2/2 cost:2} - nexthop{37 cost:1}
 03:42:24.376 [I] RouterTable---: Route table dump end
```

At this point, the direct link to `0x4c00` has a cost of 16 and cannot be used, but we have found a next-hop path through router ID `47` (`0xbc00`) with a `cost:1`, so we can send and reach `0x4c00`. I expect this to already work with how the routes are picked:
https://github.com/openthread/openthread/blob/abadeadfbbd2e71ec31bc957f2aa68c64b52ca72/src/core/thread/router_table.cpp#L459-L478

sarveshkumarv3, 2025-02-28:
Thanks 

>Regarding Address Query failures, I expect https://github.com/openthread/openthread/pull/10575 to apply rate-limiting and limit the number of attempts (I think the freshness timeout was 3 seconds?). Can you please check and confirm that this is what you observe?

That is right. The address query freshness timeout is set to 3s and hence rate-limited to 3s. 

> I know you are suggesting that maybe we allow the node to still use the direct link (with "link quality zero") to communicate with 0x4c00 (if there is no other option). This may be helpful in this [...]

The suggestion is to use the direct link only when there is no route via next hop and the outgoing link quality is still not-zero. In case we attempt transmission via direct link in such scenario and hit max retries we would anyway stop using the direct link right?