openthread/openthread #11307: [csl] `otPlatRadioEnableCsl()` API - byte order for `aExtAddr`

abtink, 2025-02-27:
Regarding the extended MAC address (`otExtAddress`), most `otPlatRadio` APIs seem to require little-endian byte order, and this is explicitly mentioned in the documentation of `otPlatRadioSetExtendedAddress()`, `otPlatRadioAddSrcMatchExtEntry()`, and `otPlatRadioClearSrcMatchExtEntry()`.

```c++
/**
 * Set the Extended Address for address filtering.
 *
 * @param[in] aInstance   The OpenThread instance structure.
 * @param[in] aExtAddress  A pointer to the IEEE 802.15.4 Extended Address stored in little-endian byte order.
 */
void otPlatRadioSetExtendedAddress(otInstance *aInstance, const otExtAddress *aExtAddress);
```

Within the OT core, the `ExtAddress` is tracked using big-endian order, and the `Radio` class ensures that the order is reversed when invoking `otPlatRadio` APIs. However, `otPlatRadioEnableCsl()` does not mention the order:

```c++
/**
 * Enable or disable CSL receiver.
 *
 * @param[in]  aInstance     The OpenThread instance structure.
 * @param[in]  aCslPeriod    CSL period, 0 for disabling CSL. CSL period is in unit of 10 symbols.
 * @param[in]  aShortAddr    The short source address of CSL receiver's peer.
 * @param[in]  aExtAddr      The extended source address of CSL receiver's peer.
 *
 * @note Platforms should use CSL peer addresses to include CSL IE when generating enhanced acks.
 *
 * @retval  OT_ERROR_NOT_IMPLEMENTED Radio driver doesn't support CSL.
 * @retval  OT_ERROR_FAILED           Other platform specific errors.
 * @retval  OT_ERROR_NONE             Successfully enabled or disabled CSL.
 */
otError otPlatRadioEnableCsl(otInstance        *aInstance,
                             uint32_t           aCslPeriod,
                             otShortAddress      aShortAddr,
                             const otExtAddress *aExtAddr);
```

This is the path where this API is called:

```c++
void Mac::UpdateCsl(void)
{
    uint16_t period  = IsCslEnabled() ? GetCslPeriod() : 0;
    uint8_t  channel = GetCslChannel() ? GetCslChannel() : mRadioChannel;

    if (mLinks.UpdateCsl(period, channel, Get<Mle::Mle>().GetParent().GetRloc16(),
                        Get<Mle::Mle>().GetParent().GetExtAddress()))
```

From this, we have `Mac::UpdateCsl()` -> `Mac::Links::UpdateCsl()` -> `Mac::SubMac::UpdateCsl()` -> `Radio::EnableCsl()` -> `otPlatRadioEnableCsl()`. There is no place in the code where the byte order is changed.

**So it seems like this API requires `aExtAddr` to use big-endian byte order.**

I want to clarify that this is indeed how it is implemented by radio platforms that implement this and update this in the documentation.

FYI: @edmont , @LuDuda can you please help confirm if this is indeed the order used by nordic radio platform implemenations? Thanks

abtink, 2025-02-28:
Well , looks like it is indeed inverted in radio platform implementation here:

- https://github.com/openthread/ot-nrf528xx/blob/dd4b07a6815a4cd562355166f3218edfab4bb473/src/src/radio.c#L1351-L1382
- https://github.com/openthread/ot-nrf528xx/blob/dd4b07a6815a4cd562355166f3218edfab4bb473/src/src/radio.c#L204-L211

edmont, 2025-03-05:
Hi @abtink, just to confirm that swap is also performed from BE to LE in NCS platform: https://github.com/nrfconnect/sdk-zephyr/blob/v3.7.99-ncs2/drivers/ieee802154/ieee802154_nrf5.c#L927

abtink, 2025-03-05:
Thanks. @edmont.

Will submit a PR later to update the documentaion and mention big-endian byte order.

Similar question applies to `otPlatRadioConfigureEnhAckProbing()` API which also gets an `otExtAddress *` and seems to be using big-endian byte order (there is no code in OT which reverses the order)

- So these APIs will expect `otExtAddress` in little-endian order: 
   - `otPlatRadioSetExtendedAddress()`, 
   - `otPlatRadioAddSrcMatchExtEntry()`, 
   - otPlatRadioClearSrcMatchExtEntry()`.
- And these will use big-endian order: 
   - `otPlatRadioEnableCsl`
   - `otPlatRadioConfigureEnhAckProbing`

closed 2025-03-27 by jwhui