openthread/openthread #11339: Test case MATN_15_ChangeOfPrimaryBBRTriggersRegistration failed after a refactor

Irving-cl, 2025-03-12:
**Describe the bug** A clear and concise description of what the bug is.
The CI failed to pass the test case MATN_15_ChangeOfPrimaryBBRTriggersRegistration with this [PR](https://github.com/openthread/ot-br-posix/pull/2727)

**To Reproduce** Information to reproduce the behavior, including:
The failed CI: https://github.com/openthread/ot-br-posix/actions/runs/13781497111/job/38550851393?pr=2755

To reproduce locally:
1.  Install a temp wireshark: Go to openthread root dir, `./script/test get_thread_wireshark`
2. Build ot simulation: Go to openthread root dir, `VIRTUAL_TIME=0 ./script/test build`
3. Build OTBR docker: `sudo LOCAL_OTBR_DIR=<the directory of ot-br-posix> ./script/test build_otbr_dockerk`
4. Run the test: `sudo VERBOSE=1 VIRTUAL_TIME=0 PACKET_VERIFICATION=1 ./script/test cert_suite tests/scripts/thread-cert/border_router/MATN/MATN_15_ChangeOfPrimaryBBRTriggersRegistration.py`

Git commit id: The lastest main branch with the [PR](https://github.com/openthread/ot-br-posix/pull/2727)

**Analysis**
The test case verifies that a secondary BBR can become a primary BBR after we shut down the primary BBR. This seems unrelated to the [change](https://github.com/openthread/ot-br-posix/pull/2727) mentioned. The change refactors the meshcop service publishing.

**How does change make a difference?**
I ran the test with and without the change. The change brings a slightly different time sequence. And the first big difference is that when we shut down the PBBR, the SBBR:
* receives a Data Response (with the change, the test failed)
* doesn't receive a Data Response (without the change, the test passed)

In the test script, we shut down the PBBR with these operations:
```
        br1.disable_backbone_router()
        br1.thread_stop()
        br1.interface_down()
```
As a result, in the failed run, we can see the PBBR state become "Removed" at time 51'':
```
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [I] BbrLeader-----: PBBR state: Removed                                                                                                                                                                                                             
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [W] BbrLocal------: !!! HandleBackboneRouterPrimaryUpdate                                                                                                                                                                                           
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [W] NetworkData---: !!! SynchronizeServerData, MleRouter IsAttached:1 mWaitingForResponse:0                                                                                                                                                         
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [W] NetworkData---: !!! SynchronizeServerData error:23                                                                                                                                                                                              
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [I] RoutingManager: Will evaluate routing policy in 00:02.367 (2367 msec)                                                                                                                                                                           
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.922 [I] NetDataPublshr: ExternalRoute fc00::/7 (state:Added) in netdata - total:2, preferred:1, desired:10                                                                                                                                              
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.925 [I] Platform------: Execute command `ipset flush otbr-ingress-allow-dst-swap` = 0                                                                                                                                                                   
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.929 [I] Platform------: Execute command `ipset flush otbr-ingress-deny-src-swap` = 0                                                                                                                                                                    
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.932 [I] Platform------: Execute command `ipset add otbr-ingress-deny-src-swap fdb2:d7ea:e54c:1::/64 -exist` = 0                                                                                                                                         
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.936 [I] Platform------: Execute command `ipset add otbr-ingress-deny-src-swap fd00:db8:0:0::/64 -exist` = 0                                                                                                                                             
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.939 [I] Platform------: Execute command `ipset add otbr-ingress-allow-dst-swap fdb2:d7ea:e54c:1::/64 -exist` = 0                                                                                                                                        
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.943 [I] Platform------: Execute command `ipset swap otbr-ingress-deny-src-swap otbr-ingress-deny-src` = 0                                                                                                                                               
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [I] Platform------: Execute command `ipset swap otbr-ingress-allow-dst-swap otbr-ingress-allow-dst` = 0                                                                                                                                             
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: [INFO]-BA------: Publish meshcop service OpenThread BorderRouter #D59E._meshcop._udp.local.                                                                                                                                                                      
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: [INFO]-BA------: Result of publish meshcop service OpenThread BorderRouter #D59E._meshcop._udp.local: OK                                                                                                                                                         
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [D] P-SpinelDrive-: Received spinel frame, flg:0x2, iid:0, tid:0, cmd:PROP_VALUE_IS, key:STREAM_RAW, len:110, rssi:-20 ...                                                                                                                          
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [D] P-SpinelDrive-: ... noise:-128, flags:0x0000, channel:11, lqi:0, timestamp:1810020068865, rxerr:0                                                                                                                                               
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [I] MeshForwarder-: Received IPv6 UDP msg, len:131, chksum:7c05, ecn:no, from:166e0a0000000003, sec:no, prio:net, rss:-20.0                                                                                                                         
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [I] MeshForwarder-:     src:[fe80:0:0:0:146e:a00:0:3]:19788                                                                                                                                                                                         
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [I] MeshForwarder-:     dst:[ff02:0:0:0:0:0:0:1]:19788                                                                                                                                                                                              
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:51.946 [I] Mle-----------: Receive Data Response (fe80:0:0:0:146e:a00:0:3)                                                                                                                                                                                 
Mar 12 08:22:28 feeec7f79c5a otbr-agent[140]: 00:00:52.092 [I] Mle-----------: Send Data Response (ff02:0:0:0:0:0:0:1)                                                                                                              [... cut ...]
As I understand, if we wait more time and `HandleCoapResponse` will be invoked with `Abort` and `mWaitingForResponse` will be false and then SBBR will become PBBR? Or we can reset `mWaitingForResponse` since the role of the device has changed (from router to leader). I did a small experiment that if we reset this in `NetworkData::Notifier::HandleNotifierEvents` RoleChanged event, the case can pass.

What do you think of this issue? @abtink 

[br2_fail.log](https://github.com/user-attachments/files/19213613/br2_fail.log)

[br2_pass.log](https://github.com/user-attachments/files/19213617/br2_pass.log)

Irving-cl, 2025-03-13:
This test case has been flaky: https://github.com/openthread/openthread/issues/10567

abtink, 2025-03-15:
Thanks @Irving-cl for analyzing this. I have been away (PTO) for a while, so I couldn't get into the details of this and comment earlier.

> As I understand, if we wait more time and HandleCoapResponse will be invoked with Abort and mWaitingForResponse will be false and then SBBR will become PBBR?

I think it would be good to understand this better.

* In `SendServerDataNotification()`, we prepare and send `PriorityConfirmablePostMessage(kUriServerData)` TMF message to the Leader ALOC using the TMF CoAP Agent.
* We use `HandleCoapResponse()` as a callback to learn when the TMF message is done (successfully or failed).
* We seem to use `mWaitingForResponse` as a way to track this. We set it after we send and clear it in the callback `HandleCoapResponse()`.

The issue seems to be that after sending such a message, the leader may get removed (go offline) and therefore not respond.

* In this case, I expect the CoAP retry mechanism itself to kick in and retry after some timeouts (and I think it uses some exponential backoff style to increase wait time after each attempt).
* Eventually, after some number of attempts, CoAP should consider the transaction failed and signal this back, invoking the callback.
* I know CoAP "transaction parameters" can be set to control the timings and number of retries (overall behavior of CoAP) `otCoapTxParameters`.
* But I assume we use the default (recommended) `otCoapTxParameters` values in `Tmf::Agent`, which I think can take some long time, and that's the reason for this failure.

> Or we can reset `mWaitingForResponse` since the role of the device has changed (from router to leader).

Your suggestion (also from the PR) is to reset `mWaitingForResponse` on role change and specifically when the device itself becomes the leader.

This sounds fine to me, but there are some things to consider:

* I think we should also cancel the earlier CoAP transmission request, along with resetting `mWaitingForResponse`.
    * Otherwise, we may get `HandleCoapResponse()` called multiple times (when the earlier transaction eventually times out).
    * We have `AbortTransaction()` in the `Coap` module, which we can use for this.
* I see in your PR that you did this when the device itself becomes the leader.
    * I think the same situation can happen if the leader is changed somewhere else in the network.
    * I am thinking whether we try to address this in a more general way to cover other cases.
    * For example, if we see a "leader age timeout" happen (which is detected in `MleRouter::HandleTimeTick`), we can signal to `NetworkData::Notifier` so it can then abort its previous CoAP transaction and reset `mWaitingForResponse`.
    * This way, it will try to register its local NetData again more quickly with the new leader, whether it is itself or some other node.
    * What do you think?

The other thing that I think would be good to check:

* When we send the `kUriServerData` TMF message, we use the Leader ALOC address as the destination.
* So, I assume if the leader gets changed, the CoAP retries should end up going to the new leader (Leader ALOC is owned by whichever node is the leader now). Therefore, we should not technically need to wait for all "CoAP" retries to fail for us to prepare and a send a new message for this to work.
* I think it would be good to check if this is happening in the case where the device itself becomes the leader (or maybe the message is dropped?).
* Is it that the BBR test happens to start at an inopportune time, and the CoAP retry time happen to be longer than the time waiting in the test script, or is there something else at play here?

Irving-cl, 2025-03-17:
@abtink 

> The other thing that I think would be good to check:
> When we send the kUriServerData TMF message, we use the Leader ALOC address as the destination.
So, I assume if the leader gets changed, the CoAP retries should end up going to the new leader (Leader ALOC is owned by whichever node is the leader now). Therefore, we should not technically need to wait for all "CoAP" retries to fail for us to prepare and a send a new message for this to work.
I think it would be good to check if this is happening in the case where the device itself becomes the leader (or maybe the message is dropped?).
Is it that the BBR test happens to start at an inopportune time, and the CoAP retry time happen to be longer than the time waiting in the test script, or is there something else at play here?

The suggestion is pivotal!

I further debugged on it and I did find a bug. What you mentioned are all correct. In the test case we reached a case where `mWaitingForResponse` is set to `true` and will never be set to `false` back. So even we wait longger in the test, the test will still fail.

Here is how it happened:
1. Leader was offline. The router (SBBR) sent a confirmable coap message.
2. As the Leader was offline, there is no reply. SBBR did re-transmission for the coap messages for 4 times.
3. When remaining re-transmission remaining is 0, SBBR hasn't become a leader and is still a router.
4. On SBBR `CoapBase::HandleRetransmissionTimer` is called and re-tranmission remaining is 0.
    * `FinalizeCoapTransaction` is called. `HandleCoapResponse` is called with result being `ResponseTimeout`.
    * `mWaitingForResponse` is temporarily set to `false`.
    * But right after that, `NetworkData::Notifier::SynchronizeServerData` is called. And it called `UpdateInconsistentData` -> `SendServerDataNotification` -> `Get<Tmf::Agent>().SendMessage` and send a confirmable coap message again!
    * `mWaitingForResponse` is set to `true` again.
    * **Key point**: `CoapBase::mRetransmissionTimer` is set to the next fire time at this point. But the code is still in the stack of `CoapBase::HandleRetransmissionTimer`.
    * At the end of `CoapBase::HandleRetransmissionTimer`, `mRetransmissionTimer.FireAt(nextTime);` is called with `nextTime` being not set.  The implementation of `FireAt` taking an unset `nextTime` stopped the timer.
 5. Now, `mWaitingForResponse` is set to `true` but `mRetransmissionTimer` is not running. We will never get `HandleCoapResponse` called.

So I think the key point here is that we send another confirmable COAP message in `CoapBase::HandleRetransmissionTimer` which set a timer but we stopped the timer at the end. A straightforward fix for this is that we only call `mRetransmissionTimer.FireAt(nextTime);` when `nextTime` is set. I think this makes sense because we don't need to explicitly stop the timer since `CoapBase::HandleRetransmissionTimer` is being called. If other callers set a timer during `CoapBase::HandleRetransmissionTimer`, we should keep it. I've verified that the small fix can make the test pass. Thoughts?

abtink, 2025-03-17:
Thanks @Irving-cl for debugging this further and your great analysis. 👍

There is indeed an issue/bug in CoAP timer re-scheduling (I'm glad we found it).

I added a comment on #11346, suggesting maybe a different fix (which addresses more situations). I submitted this: https://github.com/openthread/openthread/pull/11348 for this.

jwhui, 2025-03-18:
Resolved by #11348

closed 2025-03-18 by jwhui