openthread/openthread #11362: SED remove CSMA backoff before DataPoll

tanyanquan, 2025-03-24:
This issue is raised to consider potential power saving optimizations for SED operations (in particular rx_off_when_idle devices), specifically regarding the CSMA backoff process when a child sends DataPoll periodically.
 
We would like to ask if it's possible to remove the CSMA backoff requirement before sending out the DataPoll message so that a sleepy child does not have to wait for (potentially quite a few) CSMA backoff before sending the DataPoll message.
 
Alternatively, would it be possible to add some jitter to the sleep duration (in place of CSMA backoff) so that a sleepy child can transmit the DataPoll message upon waking up to reduce idle time?

EskoDijk, 2025-04-03:
The CSMA backoff is (I think) defined by the Thread specification, so that any change there would require a specification change.

Adding a jitter to the sleep duration would be possible, however: in what way would this help for energy saving? On average, the same number of polls will still be done.

tanyanquan, 2025-04-11:
Thanks @EskoDijk , our understanding is that the current implementation of CSMA backoff waits for a randomized period (let's say dt) of time (after waking up the device). If we can allow the device to be asleep during this randomized dt, then there could potentially be some energy savings which can accumulate to be quite significant over several DataPoll periods. 

There could be a significant difference in power consumption for a device in sleep VS in idle (in the duration of CSMA backoff), so we're considering the possibility of staying in sleep for a longer time (to include a backoff timing) before waking up the device to send DataPoll. 

For example, for each DataPoll cycle, if a device is awake for a total of 10ms to send a DataPoll, which includes 1-2ms of CSMA backoff, we're suggesting:
- remove the CSMA backoff
- instead add 1-2ms of extra sleep to compensate

This could, very roughly (ignoring differences in power consumption during tx, rx and idle), result in a 10-20% decrease in power consumption.

abtink, 2025-04-14:
@tanyanquan 

> If we can allow the device to be asleep during this randomized dt, then there could potentially be some energy savings which can accumulate to be quite significant over several DataPoll periods.

This is already supported and done in the OpenThread implementation of the CSMA backoff mechanism. You can check `SubMac::StartTimerForBackoff()`, where, based on `mRxOnWhenIdle`, the radio is put into either RX or sleep mode.

https://github.com/openthread/openthread/blob/98b88dc8f7a3a9f7300248093752455260d605df/src/core/mac/sub_mac.cpp#L476-L491

**Please note that CSMA may be handled by the radio platform layer instead (if the radio claims to provide the `OT_RADIO_CAPS_CSMA_BACKOFF` capability)**. In this case, it will be up to the radio vendor to decide whether to implement this optimization or not.

Generally, the radio platform will provide its set of capabilities in the `otPlatRadioGetCaps()` platform API. The OpenThread stack will then delegate handling of the capabilities the radio claims to support to the platform layer; otherwise, the implementation in OpenThread (`SubMac` layer) will be used instead.

I beleive most radio vendors using OpenThread stack support `OT_RADIO_CAPS_CSMA_BACKOFF`.

tanyanquan, 2025-04-15:
Hi @abtink , thank you for your detailed explanation! We understand that the current implementation of CSMA backoff allows the device to return to sleep during CSMA backoff, but in practice we've realized that the CSMA backoff delay is typically too small to allow the device to meaningfully return to sleep (i.e. the sleep and wakeup processes will take a longer time / more power than if the device stayed in idle). 

More importantly, the current implementation of CSMA backoff mechanism requires the device to wake up first before performing CSMA backoff, necessitating some additional overhead during which the device is using a higher current (compared to sleep). Even in the case where a radio vendor supports `OT_RADIO_CAPS_CSMA_BACKOFF` capability, we will need to spend some time in awake state for CSMA backoff. 

Here are some simple illustrations to demonstrate the current implementation (scenario 1):

[image]

If there were to be no CSMA backoff (scenario 2):

[image]

We're proposing that we can shift the random delay (i.e. CSMA backoff / "jitter") into the sleep duration as follows (scenario 3):

[image]

Could you let us know if this could be feasible? We've noticed that power consumption can, in practice, be reduced by around 5% to 15% when comparing scenario 1 and 2, and we expect similar power saving factors with the proposed changes.

bukepo, 2025-04-16:
That fact is that the data poll is scheduled by OpenThread core, so the platform doesn't have a chance to do the optimization. I think the proposal is not removing the CSMA backoff, but pre-compute the backoff and add it to the poll timer when scheduling the next transmission. So effectively, the CSMA backoff is done, but avoid waking up the CPU again to schedule the CSMA backoff.

Retransmissions of data polls should also do CSMA backoff, which can be handled by the platform if OT_RADIO_CAPS_TRANSMIT_RETRIES is declared.

tanyanquan, 2025-07-11:
Hi @jwhui , thanks for letting us know about the additional NCP support! I took a brief look at the list of PRs, but this issue is more related to an isolated SED's behavior, without depending on NCP. 

I'm not sure if I may have missed out any specific PR in the list - is there any new functionality which you were referring to in particular?