openthread/openthread #11371: [srp] SRP client must not retransmit SRP Update when YXDomain error received

EskoDijk, 2025-03-27:
**Describe the bug** A clear and concise description of what the bug is.
In a test run, it's observed that OT SRP client retries an SRP Update in case the SRP Registrar (server) responds with the YXDomain (6) error code, also known as "Name Exists".

In the SRP draft, it's clearly disallowed to retry with a new DNS transaction (Section 3.2.5.2): 

> In this case, the SRP requester MUST choose a new name or give up.

(Retrying by the client is still possible if the client didn't receive a response. However I presume in that case, it would retry with the same DNS transaction ID?)

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

(can be provided if needed)

**Expected behavior** A clear and concise description of what you expected to happen.

**Console/log output** If applicable, add console/log output to help explain your problem.

(PCAP file is available)

**Additional context** Add any other context about the problem here.

Related certification test is SRPC-3.3. It requires that the service instance name used in the second (follow-up) SRP Update differs from the name used in the first SRP Update.

abtink, 2025-03-27:
@EskoDijk To clarify, the idea is that this should be handled by the next layer and is not intended to be within the scope of the OpenThread SRP client implementation itself.

OT SRP client provides APIs to the next layer (app layer code) for it to start and stop it and register SRP services. The host and service names are owned and picked by the caller (e.g., `matter` code will select the instance and service names based on its own specification and rules). The OT SRP client technically cannot and should not alter the names (or any other parameter associated with a service).

There is a callback mechanism where the SRP client will signal to the next layer the outcome of registration attempts. It is up to the next layer code to decide the next action.

So, if OT SRP is asked to register certain service/host, it will continue to do so until it is explicitly told to stop (API is called to stop or remove the service registration).

There are also other existing configs in OT where, upon receiving certain response codes (and/or upon repeated regitration failures), the SRP client will try to switch and pick a different server. This is applicable if the OT SRP client "auto-start" feature is enabled, which allows it to select the server on its own based on observed NetData SRP/DNS entries, and there are more than one available entry.  See `OPENTHREAD_CONFIG_SRP_CLIENT_SWITCH_SERVER_ON_FAILURE` and its related other configs such as `OPENTHREAD_CONFIG_SRP_CLIENT_DISALLOW_SERVER_SWITCH_WITH_REGISTERED_HOST`.

**That said, I know that the next layer code is often awful at handling such errors, and I expect project integrators will mess these things up and cause more issues. I will think if it makes sense to add a certain mechanism in the OT SRP client to automatically stop registrations (on certain response code errors).**

>   Related certification test is SRPC-3.3. It requires that the service instance name used in the second (follow-up) SRP Update differs from the name used in the first SRP Update.

This is a somewhat weird test. I am not sure if we want to say the Thread stack should override `matter` (or whatever app is being used) and choose a different name on its own.

abtink, 2025-03-27:
@EskoDijk , in Thread, particularly with Matter use case, mDNS name conflicts are often self-inflicted and/or transient. These can arise from a stale registration by the same device with a different Border Router, improper/stale mDNS cache, or even inconsistent behavior of a specific mDNS library used on a BR.

So, allowing the SRP client to retry its registration (regardless of the earlier error) can be a useful and even the desired behavior. Note that we already have a backoff mechanism for retries (so they will become less frequent if failures persist). Furthermore, the next layer is always in control and is notified of each failure via a callback and can then decide to take action or allow retries to continue.

The `OPENTHREAD_CONFIG_SRP_CLIENT_SWITCH_SERVER_ON_FAILURE` configuration is particularly important for addressing such situations. Upon receiving a "Name Exists" response from the current server, it immediately triggers a server switch. This enables the device to locate the previous BR, which might still hold its prior registration causing the name conflict.

Considering these points, it makes sense to maintain the current behavior of the OT SRP client, i.e., once its API is called to register a service or host, it should continue attempting registration until explicitly instructed otherwise (its API is called to stop or remove the service). This approach appears to be the most flexible and even desired.

Generally, I don't believe we should require or specifically test this behavior for Thread devices, even if it is mentioned in the SRP RFC draft. We can consider adding a clarification regarding this in Thread specifications.

EskoDijk, 2025-03-28:
@abtink Thanks, I agree with many of the points you raised. I created SPEC-1398 to consider spec updates / clarifications for this and to direct a change in the test plans.

However, logically for a Thread stack (component) which the test is about, the default behaviour on YXDomain error should be to report to the higher layer and stop retrying after all SRP server options have been exhausted. The higher layer can then try again if it wants, in a couple of hours for example.
But if we don't trust the higher layer to do the right thing, doing it in the Thread stack layer is an option. E.g. it could provide more sensible timeouts (e.g. exponential going up to hours instead of retry every N seconds) ?

So the test was not specific to test this retry behavior, but rather the way the component test was written it expects 0 retries upon YXDomain error because there's only 1 SRP Registrar in the mesh.

abtink, 2025-04-04:
Thanks @EskoDijk.

I still think it does not make sense to implement this in the OpenThread SRP client and/or even perform such a test as part of Thread certification.

- I think retries are indeed useful and important, definitely when there are multiple servers, but also for the single-server case.
- Retries can handle transient failures caused by quirks of different mDNS implementations (we've seen so many weird behaviors).
- OpenThread SRP client retries already use an exponential backoff mechanism (starting from 1.8 seconds, up to a maximum of 1 hour with multiplication factor of 1.7).
- The next layer is always in control and can decide the next action.

> I will think if it makes sense to add a certain mechanism in the OT SRP client to automatically stop registrations (on certain response code errors).

- While saying "on a YXDomain error the SRP implementation must never send to the same server again" sounds okay and easy, implementing (and testing) this adds a non-trivial amount of complexity to the code: 
   - We need to track a new state per service; for example, when multiple services are registered, it’s possible that the first service is successfully registered, but a name conflict is reported for the second/next service(s).
   - In the case of the "switch server" feature, we need to track whether we have tried to register every conflicted service with all the available/discovered servers (this requires a new data structure to track all servers and then a state per service and server).
   - This adds code and memory size overhead for a behavior that I am not sure would help or is desired.

On a personal note, you may have noticed that my availability has been more limited in the past few months (due to multiple family health issues), and this may continue to be so in the future. I need to decide where my contribution can help most and have more impact.

EskoDijk, 2025-04-04:
@abtink 

> Retries can handle transient failures caused by quirks of different mDNS implementations (we've seen so many weird behaviors).

I think this is a very valid point. E.g. there could be stale entries stuck in a cache temporarily, which could be resolved by a retry later on. I've just seen a case where data was advertised wrongly via mDNS using a TTL of 10 seconds; and the default PTR/host record TTL is 2 minutes, and other record types 75 minutes.  So retries after these respective times would make sense (with exponential backoff) to see if the conflict is resolved, for the Matter case for example.

> implementing (and testing) this adds a non-trivial amount of complexity to the code

Agree we should not go into this direction. It's also not needed if the higher layer is in control: in principle the name can be retried when the higher layer says so and it's not violating any RFC requirements in the protocol implementation itself.

Because the higher layer may not be "aware" of the need for retry or the best retry strategy we could still keep the function to do this in the SRP client (at the Thread stack layer) to handle the transient failure case. (We then mark this as allowed deviation from draft-ietf-dnssd-srp , which will itself be anyway superseded in this aspect by draft-ietf-dnssd-advertising-proxy .)   One easier future extension to consider is making the initial retry time longer if 1) YXDomain is returned and 2) the same SRP registrar is targeted for the retry SRP Update. 

For now I will update the test plan so that retries will be correctly handled. The test itself was just a test for the SRP client component, to check that in case of name conflict it either 1) notifies the higher layer about the conflict, and/or 2) automatically picks a new name and tries to register that.   Now case 2) doesn't make much sense given that protocols like Matter pick their own unique names that don't change. So I'll exclude that from the test case.  The decision to do a test for a Thread component's SRP client in the first place was made long ago at a Thread meeting - I assume that's still useful to have?