openthread/openthread #11396: [dns] Support for SOA record in Thread Service Registry

EskoDijk, 2025-04-03:
**Describe the bug** A clear and concise description of what the bug is.

It was raised in SPEC-1389 that a SOA record is mandatory for an SRP Registrar and/or DNS authoritative server located in the same Thread Device. A client could query for the SOA record to check liveness of the server, or determine the default registration domain per the Section 3.1.1 procedure in draft-ietf-dnssd-srp.

During tests, it was found that OTBR doesn't support the SOA record.

**To Reproduce** Information to reproduce the behavior, including:

(any OTBR)

**Expected behavior** A clear and concise description of what you expected to happen.

An SRP requester could (in a role of DNS client) request a SOA record for "default.service.arpa". DNS authoritative server is expected to return it with suitable values.

Exact values are under discussion in SPEC-1389. Some initial guidance is in draft-ietf-dnssd-srp.

**Console/log output** If applicable, add console/log output to help explain your problem.
-

**Additional context** Add any other context about the problem here.
-

abtink, 2025-04-03:
>   During tests, it was found that OTBR doesn't support the SOA record.

Support for arbitrary queries (beyond PTR, SRV, TXT, AAAA, A) is being added in:

- #11357
- Along with this on the client side: #11359

   
It should be easy to add support for SOA after those are merged and I will take care of it. But to be frank, I don't think this is really useful.

If the goal is determine live-ness of BR, I would never suggest doing a DNS query (we can, of course, but would be inefficient). I would do a TMF "NetDiag" command instead.

I would personally say adding support for SRP Coder (SRP compression) is way more important than such a feature. As engineers, our job is to assess the usefulness of features and prioritize. The fact that we need to send 500-byte (even up to 1000-byte) SRP messages from Thread devices to say that it is a "Matter device" and indicate its address/port, is super inefficient.

EskoDijk, 2025-04-04:
Sorry, just discovered this is a duplicate of https://github.com/openthread/ot-br-posix/issues/2684 which was assigned to @librasungirl .  This also has some thoughts on where to implement it.

The usefulness of SOA can be discussed as part of SPEC-1389. As I understand, it's just something a DNS server and an SRP Registrar are always expected to have - even if OpenThread-based devices would normally not query for it.
This was raised because the issue blocked a cert test from succeeding; hence the (perceived) priority of this.

abtink, 2025-04-16:
PR adding support for SOA:
- https://github.com/openthread/openthread/pull/11413

jwhui, 2025-05-21:
REsolved by #11413

closed 2025-05-21 by jwhui