openthread/openthread #11464: Udp::SocketHandle::Matches filters messages when NetifId is unspecified

Irving-cl, 2025-04-30:
I found that with my PR to enable backbone router multicast forwarding feature in NCP, the CI failed in NCP border agent commissioning case. It turned out that after enabling `OPENTHREAD_CONFIG_BACKBONE_ROUTER_ENABLE`, the socket match for DtlsTransport failed:
```
bool Udp::SocketHandle::Matches(const MessageInfo &aMessageInfo) const
{
    bool matches = false;

#if OPENTHREAD_FTD && OPENTHREAD_CONFIG_BACKBONE_ROUTER_ENABLE
    VerifyOrExit(IsBackbone() == aMessageInfo.IsHostInterface());  // <--- !!! Exit here
#endif

    VerifyOrExit(GetSockName().mPort == aMessageInfo.GetSockPort());

    VerifyOrExit(aMessageInfo.GetSockAddr().IsMulticast() || GetSockName().GetAddress().IsUnspecified() ||
                 GetSockName().GetAddress() == aMessageInfo.GetSockAddr());

    // Verify source if connected socket
    if (GetPeerName().mPort != 0)
    {
        VerifyOrExit(GetPeerName().mPort == aMessageInfo.GetPeerPort());

        VerifyOrExit(GetPeerName().GetAddress().IsUnspecified() ||
                     GetPeerName().GetAddress() == aMessageInfo.GetPeerAddr());
    }

    matches = true;

exit:
    LogInfo("SocketHandle Matches:%d", matches);
    return matches;
}
```
Because the mNetifId of DtlsTransport is 'Unspecified' and `IsHostInterface` is true for the UDP message.  I think we should allow the Match to succeed when the NetifId is unspecified. Thoughts? @abtink

abtink, 2025-04-30:
Relaxing the check seems reasonable to me, though I am also concerned about potential side effects. I am not very familiar myself with all the backbone router components, and this check seems to be related to them.

One question: How is `IsHostInterface()` set on the UDP message? Is it set by `otUdpForwardReceive`?
From the code, it appears to be set either by `otUdpForwardReceive()` or within the `bbr_manager` source file

abtink, 2025-04-30:
Submitted 
- https://github.com/openthread/openthread/pull/11465

Irving-cl, 2025-05-05:
> One question: How is IsHostInterface() set on the UDP message? Is it set by otUdpForwardReceive?
From the code, it appears to be set either by otUdpForwardReceive() or within the bbr_manager source file

Yeah, in the case of BorderAgent in NCP mode, it's set by `otUdpForwardReceive`. I think it makes to relax the match here.

closed 2025-05-05 by jwhui