openthread/openthread #11475: Build on alpine linux and musl

JustAnotherSwCoderAndTester, 2025-05-06:
I think the posix CMake build should be updated to work on Alpine linux or with musl instead of glibc. I tried to accomplish this based on the commit ID 9681690fab100590566e4937cbf2d072de031ff3, but diff of relevant file shows this should also be an issue on main branch.

The anl library is added but only available if we are on a glibc system: https://github.com/openthread/openthread/blob/19203d32877461d9bc89846cc75dd5a93230a0c7/src/posix/platform/CMakeLists.txt#L184

The following should be a suitable solution.
```cmake
    FIND_LIBRARY(LIB_ANL anl)
    IF(LIB_ANL)
        target_link_libraries(openthread-posix PRIVATE anl)
    ENDIF()
``` 
This approach was used for a similar problem: [https://lists.buildroot.org/pipermail/buildroot/2020-June/273022.html](url)

This is sufficient to make the following dockerfile build and have ot-daemon and ot-ctl.
```dockerfile
FROM alpine:3.21 as cpp_build_openthread
RUN apk add --upgrade --no-cache \
	g++ \
	gcc \
	make \
	cmake \
	git \
	tar \
	xz \
	linux-headers \
	bash \
	ninja-is-really-ninja

WORKDIR /App
COPY clone ./
WORKDIR /App/build
RUN cmake -S ../ -B ./ \
	-GNinja \
	-DOT_SLAAC=ON \
	-DOT_ANYCAST_LOCATOR=ON \
	-DOT_BORDER_AGENT=OFF \
	-DOT_BORDER_AGENT_ID=ON \
	-DOT_BORDER_ROUTER=OFF \
	-DOT_CHANNEL_MANAGER=ON \
	-DOT_CHANNEL_MONITOR=ON \
	-DOT_COAP=ON \
	-DOT_COAPS=ON \
	-DOT_COAP_BLOCK=ON \
	-DOT_COAP_OBSERVE=ON \
	-DOT_COMMISSIONER=ON \
	-DOT_COMPILE_WARNING_AS_ERROR=ON \
	-DOT_COVERAGE=ON \
	-DOT_DATASET_UPDATER=ON \
	-DOT_DHCP6_CLIENT=ON \
	-DOT_DHCP6_SERVER=ON \
	-DOT_DIAGNOSTIC=ON \
	-DOT_DNSSD_SERVER=ON \
	-DOT_DNS_CLIENT=ON \
	-DOT_ECDSA=ON \
	-DOT_HISTORY_TRACKER=ON \
	-DOT_IP6_FRAGM=ON \
	-DOT_JAM_DETECTION=ON \
	-DOT_JOINER=ON \
	-DOT_LOG_LEVEL_DYNAMIC=ON \
	-DOT_MAC_FILTER=ON \
	-DOT_NEIGHBOR_DISCOVERY_AGENT=ON \
	-DOT_NETDATA_PUBLISHER=ON \
	-DOT_NETDIAG_CLIENT=ON \
	-DOT_PING_SENDER=ON \
	-DOT_RCP_RESTORATION_MAX_COUNT=2 \
	-DOT_RCP_TX_WAIT_TIME_SECS=5 \
	-DOT_REFERENCE_DEVICE=ON \
	-DOT_SERVICE=ON \
	-DOT_SNTP_CLIENT=ON \
	-DOT_SRP_CLIENT=ON \
[... 83 lines cut ...]
/App/src/posix/platform/infra_if.cpp:758:12: note: forward declaration of 'struct ot::Posix::InfraNetif::DiscoverNat64Prefix(uint32_t)::gaicb'
  758 |     struct gaicb    *reqs[1] = {nullptr};
      |            ^~~~~
/App/src/posix/platform/infra_if.cpp:773:12: error: invalid use of incomplete type 'struct ot::Posix::InfraNetif::DiscoverNat64Prefix(uint32_t)::gaicb'
  773 |     reqs[0]->ar_request = hints;
      |            ^~
/App/src/posix/platform/infra_if.cpp:758:12: note: forward declaration of 'struct ot::Posix::InfraNetif::DiscoverNat64Prefix(uint32_t)::gaicb'
  758 |     struct gaicb    *reqs[1] = {nullptr};
      |            ^~~~~
/App/src/posix/platform/infra_if.cpp:780:28: error: 'GAI_NOWAIT' was not declared in this scope
  780 |     status = getaddrinfo_a(GAI_NOWAIT, reqs, 1, &sig);
      |                            ^~~~~~~~~~
/App/src/posix/platform/infra_if.cpp:780:14: error: 'getaddrinfo_a' was not declared in this scope; did you mean 'getaddrinfo'?
  780 |     status = getaddrinfo_a(GAI_NOWAIT, reqs, 1, &sig);
      |              ^~~~~~~~~~~~~
      |              getaddrinfo
[401/797] Building CXX object src/posix/platform/CMakeFiles/openthread-posix.dir/misc.cpp.o
[402/797] Building CXX object src/posix/platform/CMakeFiles/openthread-posix.dir/daemon.cpp.o
[403/797] Building CXX object src/posix/platform/CMakeFiles/openthread-posix.dir/hdlc_interface.cpp.o
[404/797] Building CXX object src/posix/platform/CMakeFiles/openthread-posix.dir/multicast_routing.cpp.o
[405/797] Building CXX object src/posix/platform/CMakeFiles/openthread-posix.dir/netif.cpp.o
ninja: build stopped: subcommand failed.

``` 
Unfortunately, I did not find any solution for this, but having ot-daemon and ot-ctl is sufficient for me. So I thought, I post my findings here.

jwhui, 2025-05-08:
I just merged the following PR. Does this help you achieve your goals?
- https://github.com/openthread/openthread/pull/11481

JustAnotherSwCoderAndTester, 2025-05-12:
Thank you, yes this helps achieving my goals!

If libanl is not required at all (also in glibc case), removing it from the linking at all is the best solution.

closed 2025-05-12 by JustAnotherSwCoderAndTester