openthread/openthread #11491: [tcat] Update of TCAT-agent states to allow TMF activation and temporary activation

EskoDijk, 2025-05-12:
**Is your feature request related to a problem? Please describe.** A clear and concise description of what the problem is.

For TCAT, we intend to add support for temporary activation and TMF-triggered activation. This requires an update to the current TCAT-agent and BLE-Secure state mechanisms. Currently it can only be on or off. What we need is a kind of "standby" mode where are parameters and callbacks are only provided once at Start() , but TCAT Advertisements and TCAT Commissioner access isn't enabled yet.

When TMF causes a temporary activation of TCAT, the Advertisements and Commissioner access will start.

**Describe the solution you'd like** A clear and concise description of what you want to happen.

Proposal is to have these states and formal transitions between them:

* Disabled (like today)
* Standby (everything is configured, waiting for temporary activation or app-layer based always-on activation).  No planned activation yet. No TCAT Advertisements.
* Standby for timed activation (an activation of TCAT at a specified time delay has been requested via TMF. Timer counting down to that moment.) No TCAT Advertisements.
* Enabled (temporary) - based on a duration timer, will go to standby again.  If a connection is ongoing, we will wait for that connection to stop before going to standby. Sending TCAT Advertisements.
* Enabled (persistently) - app-layer has instructed TCAT to be on. Will only go to standby or disabled again driven by the app-layer. Sending TCAT Advertisements.

The TMF message activation will then be added in a separate PR.

**Describe alternatives you've considered** A clear and concise description of any alternative solutions or features you've considered.
None yet

**Additional context** Add any other context or screenshots about the feature request here.
Implementation ongoing. Ideally, for a PR we want to build it on top of #11402 .

EskoDijk, 2025-05-12:
Item can be assigned to @EskoDijk .   FYI @arnulfrupp  I'd like to build this on top of #11402 if possible.  Have a look at the design proposal, to see if it fits our needs!

arnulfrupp, 2025-06-02:
How would the suggested TCAT API be re-directed to the active transport? 

The TCAT agent is currently owned by the transport layer object (BleSecure) as a member of the class to allow for multiple instances of TCAT agent, each owned by its respective transport. The TCAT agent is currently not aware of its owning transport class. Therfore re-directing is not easy. We would need a base class for all TCAT transports and ensure a common API.

The currently TCAT CLI is practically an "BLE secure" CLI.  We need to decide how this should be in the future. Multiple APIs covering every transport separately. Or one API with a "select" command deciding to which transport it talks.
There may be differences: TCAT over CoAP over Thread may never have a need to go onto a standby mode !?

A possible designe would be using the base class for all TCAT transports as a hook for the CLI and continue using the transport as the primary entry for API calls from the  CLI.
"tcat select xxx"  would update the point to the transport if multiple exist.

arnulfrupp, 2025-06-02:
Discussed in Thread TC Commercial call June 2nd - we may consider a two step approach:

1) Add Standby functionality to BleSecure and wird this up in TCAT CLI without making other changes to close TCAT over BLE project
2) Refactor the code such that the CLI uses a common TCAT API (residing in TCAT agent or separate TcatTransport abstraction)

Transport specific Step 1 implementation should stay re-usable in Step 2 (e.g. switching BLE advertising on and off).

Step 2 may also include refactoring BleSecure such that multiple TCAT transports all use the same TCAT agent. 
We need to have clear understanding if multiple transports all share the same TCAT state (thus instance). E.g. can one be active and the other in standby (we may agree only one should be connected at a time). Do all use the same advertising variables and call-back etc.

EskoDijk, 2025-08-10:
@arnulfrupp FYI, I've now created the TCAT states PR here: https://github.com/openthread/openthread/pull/11802

Some things to note:
- there is no tcat CLI command like `tcat state` to see the current state of the TCAT agent (standby, active, stopped, ...). But the INFO log messages will show the state when it changes. This can be tested also in a simulation build with full logs enabled. So maybe the command wouldn't be required.
- some comments that use function references have now got a hashtag (#) in front of a function name to make the function easily findable by a click in the IDE. However I'm not sure if this feature works in other IDEs than mine. I was assuming it's a Doxygen feature.
- CoAP command to do timed activation of TCAT is not yet included - for a separate PR.

EskoDijk, 2025-09-04:
@arnulfrupp One of the main reasons an operator would like to disable TCAT is security. Having TCAT enabled creates a potential "backdoor" into the device, which is ok to have for some time (e.g. during a large device commissioning project - no infrastructure yet).  But once the device is operational, TCAT would need to go to standby and only enabled in specific cases:
* for example, a Thread Commissioner on the infrastructure network would send an "activate" command to a specific device, allowing a local service person to access the device for a brief time window (e.g. 30 mins) using TCAT-over-link-local-P2P-Thread.

Also when TCAT is not needed, we don't want devices responding to TCAT discovery requests. This would have some security/privacy aspects but also to avoid excessive radio messages in case there's a dense deployment of devices. E.g. the local service person wants only uncommissioned TCAT devices to respond and not the light bulbs that were already installed 2 years ago.

EskoDijk, 2025-09-04:
It is good IoT design principle to consider every open port/pin/interface as a security concern or "attack surface". Then the design that follows will keep things closed, unless there's a need.  Also per the spec (13.12.9) a vendor MAY opt to have TCAT automatically disabled after Commissioning or after attaching is successful. So the OT stack should support the choice of "keep TCAT always on" as well as "disable TCAT once joined to the network", as well as "disable TCAT once commissioned and go back to TCAT mode if I can't reach anyone for a long time" and other solutions presented in 13.12.9.

Maybe we could discuss this on Monday!

arnulfrupp, 2025-09-04:
Yes, let’s discuss at the next TC commercial call.

I didn‘t mean to say we should not consider TCAT a relevant attack vector in security analysis. I meant we should not consider TCAT standby &  remote activation a security feature because certain devices may never go into standby mode or wake up for other reasons out of control of the system administrator.