openthread/openthread #11738: Possibly inefficient default DNS query behavior?

EskoDijk, 2025-07-17:
**Describe the bug** A clear and concise description of what the bug is.
This is something I've seen in many, almost all, capture files where a OT DNS client is tested with an OTBR as DNS resolver. The goal of this issue is to determine whether it's a bug, a coincidence, or intentional design. The current approach looks quite inefficient in my view.

The situation is a DNS unicast query which doesn't generate any answer internally from the SRP Registrar, hence the Discovery Proxy is activated to send the query via mDNS on AIL and after 6 seconds of trying it should return the unicast DNS response with 0 answer records.

What the captures show is as follows:
1. Client sends DNS request
2. BR starts mDNS discovery process
3. After about ~ 6.0 seconds after step 1, the client resend the query because it didn't get any response.
4. After about ~ 0.1-0.3 seconds after step 3, the BR/server finally responds to the first query.
5. However, the BR/server also received the retransmitted query and treats this as a new query to be executed! The query ID is identical.
6. After about ~ 6.0 seconds after step 3, the BR/server sends another response (to the retransmitted query).

So it means that even in a "good flow" every DNS query is transmitted 2x, processed 2x, and responded 2x in case the answer is 0 records.

If the client would wait e.g. 6.5 or 7 seconds instead of 6.0 seconds before sending the retry, it would already be much more efficient I think.

Not sure if this design is intentional in some way, or just coincidence. It looks like we can improve here.

**To Reproduce** Information to reproduce the behavior, including:

1. Git commit id
2. IEEE 802.15.4 hardware platform
3. Build steps
4. Network topology

**Expected behavior** A clear and concise description of what you expected to happen.
Query is only sent once by client - client either retries earlier (e.g. after 3 seconds), or later than 6 seconds (because it knows the Discovery Proxy takes 6 seconds at most), but *not* exactly 6 seconds which is the worst of all cases.

**Console/log output** If applicable, add console/log output to help explain your problem.
(Can maybe be sent on request.)

**Additional context** Add any other context about the problem here.
occurs in several 1.3/1.4 cert tests cases.

abtink, 2025-07-18:
Thanks @EskoDijk .

The response timeout is configurable per query. See `mResponseTimeout` field:
```
/**
 * Represents a DNS query configuration.
 *
 * Any of the fields in this structure can be set to zero to indicate that it is not specified. How the unspecified
 * fields are treated is determined by the function which uses the instance of `otDnsQueryConfig`.
 */
typedef struct otDnsQueryConfig
{
    otSockAddr          mServerSockAddr;  ///< Server address (IPv6 addr/port). All zero or zero port for unspecified.
    uint32_t            mResponseTimeout; ///< Wait time (in msec) to rx response. Zero indicates unspecified value.
    uint8_t             mMaxTxAttempts;   ///< Maximum tx attempts before reporting failure. Zero for unspecified value.
    otDnsRecursionFlag  mRecursionFlag;   ///< Indicates whether the server can resolve the query recursively or not.
    otDnsNat64Mode      mNat64Mode;       ///< Allow/Disallow NAT64 address translation during address resolution.
    otDnsServiceMode    mServiceMode;     ///< Determines which records to query during service resolution.
    otDnsTransportProto mTransportProto;  ///< Select default transport protocol.
} otDnsQueryConfig;
```

If not specified the default is used which is
```c++
/**
 * @def OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_RESPONSE_TIMEOUT
 *
 * Specifies the default wait time that DNS client waits for a response from server (in milliseconds).
 */
#ifndef OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_RESPONSE_TIMEOUT
#define OPENTHREAD_CONFIG_DNS_CLIENT_DEFAULT_RESPONSE_TIMEOUT 6000
#endif
```

We can change the default to 7 seconds.

abtink, 2025-07-18:
@EskoDijk , regarding the 6-second discovery proxy timeout, do you happen to know where does it come from? is it specified somewhere? I do see it being defined here in OT as `6000`.
```c++
/**
 * @def OPENTHREAD_CONFIG_DNSSD_QUERY_TIMEOUT
 *
 * Specifies the default wait time that DNS-SD Server waits for a query response (e.g. from Discovery Proxy).
 */
#ifndef OPENTHREAD_CONFIG_DNSSD_QUERY_TIMEOUT
#define OPENTHREAD_CONFIG_DNSSD_QUERY_TIMEOUT 6000
#endif
```

abtink, 2025-07-18:
Submitted 
- https://github.com/openthread/openthread/pull/11741

EskoDijk, 2025-07-18:
@abtink The 6 seconds is in RFC 8766, 5.6 first bullet:

> If after six seconds no relevant Multicast DNS answers have been received, cancel the mDNS query and return a negative response to the remote client.

closed 2025-07-22 by jwhui